betaflight/betaflight-configurator #4508: SRXL2 Support - Missing USE_SERIALRX_SRXL2 option on compilations

bouviervj, 2025-06-09:
### Describe the bug

I'm using the Spektrum SRXL2 protocols on might flight controller, and when selecting the option SPEKTRUM for the radio protocol, the firmware that is downloaded doesn't come with the compilation option USE_SERIALRX_SRXL2 selected, despite indicating that I want SRXL support for my firmware. Is this a SRXL/SRXL2 mismatch on the options ?
When compiling with all proper options, I'm able to build a correct firmware manually.

### To Reproduce

Select radio protocol 'SPEKTRUM' the image comes back without the option USE_SERIALRX_SRXL2 (build logs):
`make JHEF7DUAL EXTRA_FLAGS="-D'BUILD_KEY=92c08efa7a2297178ef7634facd96a03' -D'RELEASE_NAME=4.5.2' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_SPEKTRUM -DUSE_TELEMETRY -DUSE_TELEMETRY_SRXL -DUSE_VTX"`

### Expected behavior

The build logs shall contain the option USE_SERIALRX_SRXL2 :
`make JHEF7DUAL EXTRA_FLAGS="-D'BUILD_KEY=92c08efa7a2297178ef7634facd96a03' -D'RELEASE_NAME=4.5.2' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_SPEKTRUM -DUSE_SERIALRX_SRXL2 -DUSE_TELEMETRY -DUSE_TELEMETRY_SRXL -DUSE_VTX`

### Configurator version

10.10.0

### Flight controller configuration

```txt

```

haslinghuis, 2025-06-09:
The root cause of your issue is that when you select the "SPEKTRUM" radio protocol in the configurator, only the [USE_SERIALRX_SPEKTRUM](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/) build option is sent to the cloud build, not

[USE_SERIALRX_SRXL2](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/). This is because the configurator's firmware flasher tab does not associate "SPEKTRUM" with SRXL2, and the build request only includes the selected protocol's value.

Why this happens:

    In [firmware_flasher.js](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/), the build request for radio protocols only pushes the selected value(s) from the radio protocol dropdown.
    The value for "SPEKTRUM" is 

[USE_SERIALRX_SPEKTRUM](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/), not

    [USE_SERIALRX_SRXL2](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/).
    There is a separate entry for "SPEKTRUM SRXL2" in the radio protocol options, but selecting "SPEKTRUM" alone will not enable SRXL2 support.

How to fix:

    If you want SRXL2 support, you must select the "SPEKTRUM SRXL2" protocol in the configurator, not just "SPEKTRUM".
    If the UI does not make this clear, or if "SPEKTRUM SRXL2" is missing, the options list in the configurator should be updated to clarify the difference or always include both options.

Summary:
This is not a bug in the build system, but a mismatch in protocol selection. "SPEKTRUM" and "SPEKTRUM SRXL2" are different options, and only the latter will enable
[USE_SERIALRX_SRXL2](https://vigilant-invention-qjvwrq5v9jh69r5.github.dev/) in the build.

closed 2025-06-09 by haslinghuis

bouviervj, 2025-06-09:
Thks, the amount of time I spent on this ...  At least I know how to compile the firmware now.