betaflight/betaflight-configurator #4551: Does not support 4.6.0?

HalfChrome1, 2025-07-24:
### Describe the bug

I'm attempting to connect my Mobula8 HD but it says that 10.10.0 (C97deaf) does not support firmware 4.6.0. It suggests I download 10.10.0. Seems like a repetitive loop. What do I do here?

[image]

### To Reproduce

Plug in quad, attempt to connect to BF.

### Expected behavior

Use BF

### Configurator version

10.10.0 (c97deaf)

### Flight controller configuration

```txt

```

haslinghuis, 2025-07-24:
For 4.6 use `app.betaflight.com` in chromium based browser

closed 2025-07-24 by haslinghuis

HalfChrome1, 2025-07-24:
Thanks. Is that just the way to go now, unless I'm out in the field?