betaflight/betaflight-configurator #4555: Regression: WebSockets appear to be broken after commmit 5ef9acb

TrevorSundberg, 2025-07-26:
### Describe the bug

Commit in question:
https://github.com/betaflight/betaflight-configurator/commit/5ef9acb440cdd2572ed41db55eccf98b5471cc43

After this commit, the latest version of https://app.betaflight.com/ can no longer connect to WebSockets. In manual connection mode `ws://127.0.0.1:6761` worked previously to connect to WebSocket endpoints, but in the latest version `11.0.0 (04ea058b)` it always reports:

[image]

### To Reproduce

Enable manual connection mode on https://app.betaflight.com/, run SITL and websockify as mentioned here:
https://betaflight.com/docs/development/SITL

Attempt to connect to `ws://127.0.0.1:6761` and notice that it does not connect.

### Expected behavior

For the configurator to connect as it was in previous versions.

### Configurator version

11.0.0 (04ea058b)

### Flight controller configuration

```txt

```

closed 2025-07-26 by VitroidFPV

Pancronos, 2025-07-30:
Getting same message

[image]

Pancronos, 2025-07-30:
It seems that if you open app browser app, then locaal app throw this message and viceversa.

haslinghuis, 2025-07-30:
Even using local build and online build and trying to connect both (using webserial) results in 

[image]

We should try to capture the error instead of this generic error. Created PR 4560.