betaflight/betaflight-configurator #4589: PID Tuning Tab: Throttle Curve UI does not reflect actual throttle curve in FC

pilmatFPV, 2025-09-05:
### Describe the bug

The Throttle Curve displayed does not reflect the setting. When a mid point is set, it is shown as an offset along the X axis. A hover point is similar along the Y axis. With mid point set at 0.30 for example, this is the displayed graph:
[image]

The curve as desired is accurate within the flight controller, tested with throttle applied and throttle increases from above zero input.

### To Reproduce

Merely set a mid point other zero or a hover point other than zero.
[image]
[image]

### Expected behavior

Throttle curve in UI to reflect desired setting.

### Configurator version

4.6.0

### Flight controller configuration

```txt
batch start

board_name HDZERO_AIO15
manufacturer_id HDZO

# name: M8 Freestyle HD

# resources
resource I2C_SCL 2 C04
resource I2C_SDA 2 F00
resource LED 1 F01

# timer
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)
timer A00 AF1
# pin A00: TIM2 CH1 (AF1)

# dma
dma SPI_SDO 2 9
# SPI_SDO 2: DMA2 Channel 2 Request 13
dma SPI_SDI 2 8
# SPI_SDI 2: DMA2 Channel 1 Request 12
dma SPI_TX 2 9
# SPI_TX 2: DMA2 Channel 2 Request 13
dma SPI_RX 2 8
# SPI_RX 2: DMA2 Channel 1 Request 12
dma ADC 1 6
# ADC 1: DMA1 Channel 7 Request 5
dma TIMUP 1 NONE
dma TIMUP 3 NONE
dma TIMUP 4 NONE
dma TIMUP 5 NONE
dma TIMUP 8 NONE
dma pin A03 4
# pin A03: DMA1 Channel 5 Request 75
dma pin A02 3
# pin A02: DMA1 Channel 4 Request 74
dma pin A01 2
# pin A01: DMA1 Channel 3 Request 57
dma pin A00 1
# pin A00: DMA1 Channel 2 Request 56
dma pin A08 9
# pin A08: DMA2 Channel 2 Request 42

# feature
feature -AIRMODE
feature OSD

# serial
serial UART1 131073 115200 57600 0 115200

# aux
aux 0 0 0 1725 2050 0 0
aux 1 1 1 1725 2050 0 0
aux 2 28 2 975 1375 0 0
aux 3 35 3 1700 2050 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM     0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B CUSTOM     0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E CUSTOM  5705    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM     0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_trim_roll = 2
set acc_calibration = 178,-55,61,1
set max_check = 1950
set motor_idle = 850
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set motor_output_reordering = 1,3,0,2,4,5,6,7
set align_board_yaw = 135
set vbat_max_cell_voltage = 440
set yaw_motors_reversed = ON
set pid_process_denom = 2
set osd_vbat_pos = 450
set osd_rssi_pos = 1548
set osd_link_quality_pos = 3176
set osd_rssi_dbm_pos = 3144
set osd_tim_2_pos = 3590
set osd_flymode_pos = 1516
set osd_throttle_pos = 2571
set osd_vtx_channel_pos = 3109
set osd_crosshairs_pos = 2361
set osd_current_pos = 514
set osd_craft_name_pos = 2057
set osd_warnings_pos = 2485
set osd_avg_cell_voltage_pos = 2584
set osd_disarmed_pos = 2230
set osd_core_temp_pos = 1580
set osd_stat_bitmask = 1828
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set gyro_1_sensor_align = CW90FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900
set craft_name = M8 Freestyle HD
set pilot_name = pilmat

profile 0

# profile 0
set p_pitch = 48
set i_pitch = 86
set d_pitch = 30
set f_pitch = 118
set p_roll = 46
set i_roll = 82
set d_roll = 27
set f_roll = 113
set p_yaw = 46
set i_yaw = 82
set f_yaw = 113
set d_max_roll = 36
set d_max_pitch = 41
set motor_output_limit = 90
set simplified_master_multiplier = 90
set simplified_pi_gain = 115
set simplified_feedforward_gain = 105

rateprofile 2

# rateprofile 2
set thr_mid = 30
set thr_expo = 30
set roll_rc_rate = 18
set pitch_rc_rate = 18
set yaw_rc_rate = 21
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 81
set throttle_limit_percent = 90

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

It appears to only be a UI issue as flight is as expected.

DavidAnson, 2025-09-09:
@nerdCopter, is that present by default? What about the difference between observed behavior and the graph? User error?

DavidAnson, 2025-09-09:
In cases where there are direct translations, is there any reason (other than resourcing!) not to provide this migration functionality? I'll note that the Configurator supports multiple ways of defining rates and I think it even converts between them automatically.